mainsail-crew/mainsail-config #15: add unload/load Filament ?

wildwick69, 2023-03-20:
wouldn't it make sense to include unload/load filament here as well so all important procedures except print_start and print_end would be covered in the same place?

zellneralex, 2023-04-14:
Unload/load filament is heavily printer depending. To make that somehow general working you would need a lot of parameters what makes it extrem hard to setup. In the most cases you are much faster to write that macro dedicated for your own printer.

For M600 many people want different park position or already include the filament unloaded and tip forming. We have a simple example in the config.
```
## Use the PAUSE macro direct in your M600:
##  e.g. with a different park position front left and a minimal height of 50 
##    [gcode_macro M600]
##    description: Filament change
##    gcode: PAUSE X=10 Y=10 Z_MIN=50
##  Z_MIN will park the toolhead at a minimum of 50 mm above to bed to make it easier for you to swap filament.
```
Also the PAUSE macro is written in a way that you could use it with other X:Y and a minimum Z distance so that adding a M600 is coping that 3 lines and modifying 3 numbers.

wildwick69, 2023-04-14:
Hi, many thanks for your answer!!

 

Think you are right, so I already implemented the variables to my load/unload/m600 + physical buttons like this:

 

[gcode_button UNL_FILA]

pin: ^!PG12

press_gcode:

    {% if printer.idle_timeout.state == 'Printing' %}                                    ; if printing      

       PAUSE Z_MIN=50                                                                    ; goto colour change position with minimum Z 50mm or Z+2mm if above    

    {% endif %}

       UNLOAD_FILAMENT

 

[gcode_button L_FILA]

pin: ^!PG13

press_gcode:

    {% if printer.idle_timeout.state != 'Printing' or printer.pause_resume.is_paused %}  ; load only if not currently prining or print is paused

      LOAD_FILAMENT

    {% endif %}

 

[gcode_macro UNLOAD_FILAMENT]

description:Unload Filament

gcode:

    {% set change_x = printer['gcode_macro _CLIENT_VARIABLE']['custom_park_x'] %}        ; set colour change position -> mainsail-configs Variable

    {% set change_y = printer['gcode_macro _CLIENT_VARIABLE']['custom_park_y'] %}        ; set colour change position -> mainsail-configs Variable

    {% set hotendtemp = params.HOTEND|default(230)|int %}

    SAVE_GCODE_STATE NAME=unload_state

    {% if printer.pause_resume.is_paused == False %}                            ; PAUSE active? if not then normal unload with position, heatup, forming tip

      M104 S{hotendtemp}                                                        ; set & continue hotend temp, default to 230

      {% if printer.toolhead.homed_axes != 'xyz' %}                             ; Home if not already homed

        M117 Homing...

        STATUS_HOMING                                                           ; LEDs show homing

        G28                                                                     ; Alphaville: Lassie come home, come home lalala

     {% endif %}

      G1 Y{change_y} X{change_x} Z50 F6000                                      ; move toolhead to custom Park Position

      M117 Heating

[... 170 lines cut ...]
      STATUS_PRINTING

      RESTORE_GCODE_STATE NAME=load_state

      RESUME

    {% endif %}

 

 

[gcode_macro M600]description: Filament change

gcode: 

    PAUSE Z_MIN=50                                                              ; goto colour change position with minimum Z 50mm or Z+2mm if above

    UNLOAD_FILAMENT

 

 

Lg, Wick

closed 2023-04-17 by zellneralex