mainsail-crew/mainsail-config #18: Adding to macros in client variables

ksummers92, 2023-07-08:
Maybe I am missing it somewhere, but I would like to add some gcode commands to the `CANCEL_PRINT` macro to run things like stopping a heat soak or retracting filament for a MMU setup (ERCF_EJECT for example). Is there a way to edit the macros without making it "dirty" to the moonraker updater? Or (I know this anti-thesis to the entire point of this config) is there a way to opt-out of the macros altogether so that they do not interfere with custom macros?

zellneralex, 2023-07-17:
There is no real solution to your problem. If you need to customize it the only feasible way is to make a copy of it and place it in your config folder. From that moment you can edit it as you want but that means you loose the auto update and need to merge features and bugfixes by hand.
I am sure that is not what you want to hear but see our macros as the "I need to fit all printers and help the beginner"

"Opt out" is also quite easy simple remove [include mainsail.cfg] from your config and use your own macros. BTW that is what I do from the beginning.

zellneralex, 2023-07-30:
I close it. I currently see no practical solution to add the possibility for user macros inserted in the profided macros

closed 2023-07-30 by zellneralex