mainsail-crew/mainsail-config #22: Heat nozzle to saved temp in resume if target lower than min extrude temp

alfons-14, 2023-09-03:
after last update we lost on resume at least 600g of filament (at least 1 day of print and night at pause) beacuse printer lost IDLE state but extruder was cold

I think that an additional check layer in resume macro to prevent cancellation of print would be great

here an example:

{% if printer.pause_resume.is_paused %}
    {% set extruder = printer[printer.toolhead.extruder] %}
    {% if extruder.target <= printer.configfile.settings[
                             printer.toolhead.extruder].min_temp
                             | float + 0.5 %}
      M109 S{saved_extruder_temp}
    {% endif %}
  {% endif %}

zellneralex, 2023-10-17:
I am not sure why you have that issue, and need more input. 
Sure I can aid band aid after band aid but I want to understand your issue so can you send me a fresh klipper.log so that I understand your config.
Normally there is no way that you loose your idle state without doing stuff like extra movements on the head. 
Also your check would not really help as the min temp is in most cases to less to print so that the object would be also lost in that situatation

zellneralex, 2023-11-01:
With the help of a other user I might now understand your problem. A solution for it will be in the nest commit.
It contains 2 things:
1) you add a call in your idle_timeout
2) A general safeguard if you want to resume a print with
  a) a cold extruder
  b) a triggered runout sensor

zellneralex, 2023-11-05:
https://github.com/mainsail-crew/mainsail-config/pull/26 should fix your problem

closed 2023-11-05 by zellneralex