mainsail-crew/mainsail-config #29: Bed temperature not restored after Pause

trophysoft, 2023-11-19:
### Mainsail Version:

2.8.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

If the print is paused and then resumed, Mainsail restores the nozzle temperature but not the bed temperature.

### What did you expect to happen instead?

Restore the bed temperature as well.

### How to reproduce this bug?

Pause the print, wait for a while to drop the bed temperature and then resume the print.

zellneralex, 2023-11-21:
@trophysoft so have you changed your idle timeout or can you explain me how you insure that the print says on the bed with a cold bed?

zellneralex, 2023-12-28:
No answer for a month...

closed 2023-12-28 by zellneralex