mainsail-crew/mainsail-config #30: mainsail.cfg comment typos

twshield, 2023-11-25:
In header of mainsail.cfg:
```
#variable_user_pause_macro : ""    ; Everything insight the "" will be executed after the klipper base pause (PAUSE_BASE) function
#variable_user_resume_macro: ""    ; Everything insight the "" will be executed before the klipper base resume (RESUME_BASE) function
#variable_user_cancel_macro: ""    ; Everything insight the "" will be executed before the klipper base cancel (CANCEL_PRINT_BASE) function
```
three instances of `insight` should be `inside`

 Thanks for all your hard work on this.

twshield, 2023-11-25:
Found another one, this time in README.md:

> 
> To remove the "Pause at next Layer" simple send
> 
> ```txt
> SET_PAUSE_AT_LAYER ENABLE=0
> ```

where `SET_PAUSE_AT_LAYER` should be `SET_PAUSE_NEXT_LAYER`, and "simple" should be "simply" (here and in next bit on SET_PAUSE_AT_LAYER (where this was copied from ;).

pedrolamas, 2023-11-25:
Not sure you are aware, but you can propose modifications directly here on GitHub!

Just find the file you want to change, click the small pencil to create a fork in your account, edit the file, and submit a pull request with the changes, all with just a few clicks!

[image]

twshield, 2023-11-25:
I did know, but did not think typos rated a fork.  Had not tried the pencil shortcut, so I'll try that next time.

zellneralex, 2023-12-28:
https://github.com/mainsail-crew/mainsail-config/releases/tag/V1.2.0

closed 2023-12-28 by zellneralex