mainsail-crew/mainsail-config #32: mainsail.cfg unicode symbol issue

UserDmitry, 2024-02-08:
Klipper report an issue "Unable to open \*path*\mainsail.cfg file". In klippy.log there is a error:
**UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 7558: ordinal not in range(128)
Config error
Traceback (most recent call last):
  File "/home/sapphire/klipper/klippy/configfile.py", line 159, in _read_config_file
    data = f.read()
  File "/home/sapphire/klippy-env/lib/python3.6/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 7558: ordinal not in range(128)**

This byte appears at str. number 137:
 **# we need to use the unicode (\u00B0) for the ° as py2 env's would throw an error otherwise** 

Unicode symbol in this comment " ° " cause this error.

pedrolamas, 2024-02-09:
I've tried to reproduce this issue on a Docker container running Python 2.7.18 and it worked fine...

@UserDmitry can you please upload a full klippy.log so we can check the issue directly?

UserDmitry, 2024-02-09:
Yes, of course. 
[klippy.log](https://github.com/mainsail-crew/mainsail-config/files/14220587/klippy.log)

pedrolamas, 2024-02-09:
Looking at the log I can see Klipper is using python 3.6.9, so that might be the problem as AFAIK, unicode support was added in 3.9.

Nevertheless, I do think we should fix this issue here as it is such a trivial fix.

@zellneralex any comments on this?

zellneralex, 2024-02-09:
I really love it, I am only adding that comment so that people understand the cryptic Unicode usage here. I will change the comment in the next days. 

Currently I am on a business trip.