mainsail-crew/mainsail-config #34: Imbalance between _CLIENT_RETRACT and _CLIENT_EXTRUDE

sk8board, 2024-07-15:
### Problem Description

When using Firmware Retract with mainsail.cfg, Pause and Resume, the extruder will over extrude during Resume.

### Cause

The mainsail.cfg macro, _CLIENT_EXTRUDE, supports Firmware Retract while _CLIENT_RETRACT does not support Firmware Retract.

### Example

If Firmware Retract is set to 10mm retract but the _CLIENT_VARIABLE is set to default 1mm, then _CLIENT_EXTRUDE will over-extrude by 9mm because _CLIENT_RETRACT does not support Firmware Retract.

### Recommended Solution

Add Firmware Retract to _CLIENT_RETRACT

_CLIENT_VARIABLE
[image]

_CLIENT_EXTRUDE
[image]

_CLIENT_RETRACT
[image]

flowerysong, 2024-07-15:
As you can see in your screenshot `_CLIENT_RETRACT` calls `_CLIENT_EXTRUDE`, so it supports exactly the same things that `_CLIENT_EXTRUDE` does.

sk8board, 2024-07-15:
Please note that I understand lines 268 through 273 should be used when retracting, but there is a problem of severe over extrusion during Resume.

sk8board, 2024-07-15:
I have verified `[firmware_retraction]` has `unretract_extra_length` set to zero.

meteyou, 2024-07-15:
@sk8board plus close this issue, when all questions are answered.

sk8board, 2024-07-15:
I believe I know what is my problem, let me explain.

- At the end of a PAUSE, I have a macro drive the nozzle to the bed, Z=0.000, to prevent the nozzle from oozing while paused. 
- While the nozzle is at the bed, filament is oozing to the bottom of the nozzle but cannot escape.  
- This results in an air pocket between the oozed filament at the bottom and the solid filament above.
- When un-retracting G11 (15mm), the air pocket does not allow the filament to fully unretract which causes the extruder gears to skip on the filament.

Does this sound reasonable, or could there be a problem with the PAUSE and RESUME macros?

meteyou, 2024-07-15:
@sk8board if you have only issues with your gcode, pls use your discord server or the discussions section on github. the issue tracker is no place for help. we try to track real bugs and features with it.

closed 2024-07-15 by meteyou

sk8board, 2024-07-15:
@meteyou This is not a problem with my gcode.

meteyou, 2024-07-15:
i read it and your issue has nothing todo with the mainsail-config macro. its just your usecase and your gcode. so pls stop spamming this issue tracker here...